ImperialSpaceSociety/LoRaMac-node #214: Will the ABP credentials expire(dev_addr etc)

MedadRufus, 2021-10-01:
Find out out the mechanism for the end node to detect server network change. its a mac command periodically sent

MedadRufus, 2021-10-02:
Its quite unlikely that it will expire. But as a plan b, consider sending a special downlink to tracker to update the keys. This can be done via either the TTN or Helium network. Hopefully keys on both networks don't expire at the same time.

MedadRufus, 2021-10-02:
I have thought about the credentials expiring problem. I don't think it will for a very long time. Or else, many devices will simply become bricks as they won't know that the credentials are expired, and will tx on the expired ones.

MedadRufus, 2021-10-02:
In theory, the OTAA end device is meant to periodically send a LinkCheckReq to check if its connected to the network. If notresponse(LinkCheckAns) from server(via a downlink), it is meant to join again. This is simply not going to work on the balloon tracker, as its going to spend most of its flight out of range. Then it will keep trying to join. I don't want it to do many joins!(or any at all!) as it requires a downlink.

MedadRufus, 2021-10-02:
The things network recommends doing an OTAA join in the factory, then persist all the keys for the lifetime of the device, suggesting that it will not ever expire.

Read it in the white paper: https://www.thethingsindustries.com/docs/devices/abp-vs-otaa/

MedadRufus, 2021-10-02:
I am hoping the same logic can be used on the helium network. I am hoping the session keys(`DevAddr`, `NwkSKey` and `AppSKey`) will be persisted on the server side for a long time(at least 1 year).

pimente, 2021-10-02:
Maybe we could ask the guys from Helium referring TTN's white paper above?

MedadRufus, 2021-10-02:
The question to ask is: What is the lifetime of the session keys(`DevAddr`, `NwkSKey` and `AppSKey`)? As a first step, we can look at the source code for the helium network, which is Open Source. Things to spot are if there is expiry logic in there. I highly doubt there is.

htdvisser, 2021-10-08:
Hi guys,

No worries.

The Things Network still does backups, so if our database servers burn down, we can restore them from those backups.

The Things Network doesn't have an expiration time on sessions, and if we would decide to add expiration in the future, it would definitely be after _at least_ a year of inactivity.

You can of course also just keep track of your DevEUIs, AppKeys, DevAddrs, NwkSKeys, AppSKeys, etc. yourself. That way you can always re-register your device on The Things Network (or any other LoRaWAN network).

Good luck with your project!

MedadRufus, 2021-10-14:
The ability to change the keys with a special downlink I think mitigates the impact of this problem. The new keys can be sent up via either network: The Things Network or the Helium Network. So if the keys expire on one network, the keys can be sent up via the other network. 
This feature has been implemented in #250

MedadRufus, 2021-12-14:
After some long term testing, I have discovered that after around 2 weeks of inactivity, helium network session keys become invalid; I have to rejoin to get new session keys. Major issue. This is not an issue on The Things Network.

@pimente have you run into this issue?

MedadRufus, 2021-12-14:
The issue appears intermittent though. In my implementation, I cycle through 10 device registration credentials, but I discovered that packets sent on credentials from one of the devices would not appear on console. All other 9 device registrations work.

[image]

MedadRufus, 2021-12-14:
I am not sure if there is an expiration date built into the ABP credentials.

pimente, 2021-12-15:
I have not done extensive tests because my reception on the Helium network on the ground is not very reliable.

I have sent a balloon on the Helium network with intermediate success because the Helium network [crashed](https://status.helium.com/incidents/zl82xrgl9p30) in the middle of the flight. The balloon disappeared after 6 days. I don't know why. Probably icing or maybe expired credentials...

According to your observations, does it expire after ~2 weeks of inactivity or two weeks on the ABP credentials? 
If it is after 2 weeks of inactivity, an easy workaround would be to send a dummy packet from the ground once every week from the same credentials to make sure they don't expire.

MedadRufus, 2021-12-15:
> According to your observations, does it expire after ~2 weeks of inactivity or two weeks on the ABP credentials?
It happens after 2 weeks of inactivity. If it continuously transmits every few minutes, there is no issue.

I do like the workaround of sending a dummy packet.

closed 2021-12-24 by MedadRufus